AndreasAmMueller/Modbus #1: ModbusClient.ReadHoldingRegisters blocks forever for non-existing devices

ygoe, 2018-08-01:
ReadHoldingRegisters calls SendRequest with reads from NetworkStream with the ReadAsync method. If no reply is sent by the Modbus server, this read operation never returns. The ReceiveTimeout setting is ignored. According to other sources, this is a limitation of the ReadAsync method.

Maybe it helps to wait for NetworkStream.DataAvailable, but that property needs to be polled. A timeout could be added to this polling loop.

A better solution is probably to use a distinct thread/Task that does nothing but reading from the network stream and then signal any new data as it arrives. Callers could then wait for that signal and decide not to wait any longer. This won't break the Receive method and the entire connection for subsequent reads.

I can't test this because I can't get the code to work as local assemblies without NuGet.

ygoe, 2018-08-01:
This works in a demo application: (ModbusClient.cs line 856)

```csharp
  var buffer = new byte[6];
+ int waited = 0;
+ while (!stream.DataAvailable)
+ {
+     if (waited > ReceiveTimeout)
+         return new Response(new byte[] { 0, 0, 0, 0, 0, 0 });
+     await Task.Delay(10);
+     waited += 10;
+ }
  var count = await stream.ReadAsync(buffer, 0, buffer.Length);
```

The same might be done for the second ReadAsync call in that method as well.

ygoe, 2018-08-03:
May I propose these classes as a basis for an asynchronous implementation: https://github.com/ygoe/AsyncTcpClient :-)

AndreasAmMueller, 2019-01-14:
27b3567 should close this issue. Please verify.

AndreasAmMueller, 2019-03-01:
Close - Reopen if not solved.

closed 2019-03-01 by AndreasAmMueller