AndreasAmMueller/Modbus #3: ModbusClient.WriteCoils modbus function #6 not working - missing Byte Count field

ajdovgnome, 2019-01-12:
ModbusClient.WriteCoils does not work correctly (yields mixed results) . After reviewing source code and actual bytes sent over modbus it was found that there is missing Byte Count in Modbus request ADU as specified in http://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0a.pdf and http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf .

Temporary fix (or possibly a solution to this) is to initialize new DataBuffer just after for loop (at line 615 in ModbusClient.cs ) and populate it, starting with Byte Count (presenting length of coilBytes) and proceding with coilBytes values:

```csharp
var data = new DataBuffer(coilBytes.Length + 1);
data.SetByte(0, (byte)coilBytes.Length);
data.SetBytes(1, coilBytes);
 ```

And assigning it to Data property of the request.

```csharp
                var request = new Request
                {
                    DeviceId = deviceId,
                    Function = FunctionCode.WriteMultipleCoils,
                    Address = firstAddress,
                    Count = (ushort)orderedList.Count,
                    Data = data // New
                    //Data = new DataBuffer(coilBytes) // Removed
                };
```

This seems to work. 
(to add: WriteRegisters function works correctly as it has Byte Count field)

AndreasAmMueller, 2019-01-14:
Thank you for the feedback, I'll check it while writing UnitTests (my task in my rare spare time).

AndreasAmMueller, 2019-01-14:
Function now works properly.

closed 2019-01-14 by AndreasAmMueller

ajdovgnome, 2019-01-14:
Changed the title as it had and error: WriteCoils is modbus function #15 (not #6).

Cant see any relevant changes to this issue in latest commit. Will test it when time allows.

AndreasAmMueller, 2019-01-14:
https://github.com/AndreasAmMueller/Modbus/blob/27b356723aea3b6344498d2996c0eb7049229622/src/Modbus.Tcp/Protocol/Request.cs#L123

That is your magic line ;-)