AndreasAmMueller/Modbus #6: Cannot access a disposed object on last connection

underjack, 2019-06-06:
I have a TCPClient connection established, and a timer that once a second calls the following:

List<Register> regs = await modbusTCP.ReadHoldingRegisters(1, 0, 24);

This works great, unless the other end of the connection disappears (I unplug the cable from the ModbusTCP server).  Then I get an exception with the following details:

```
System.ObjectDisposedException
  HResult=0x80131622
  Message=Cannot access a disposed object.
Object name: 'System.Net.Sockets.NetworkStream'.
  Source=System
  StackTrace:
   at System.Net.Sockets.NetworkStream.EndRead(IAsyncResult asyncResult)
   at System.IO.Stream.<>c.<BeginEndReadAsync>b__43_1(Stream stream, IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncTrimPromise`1.Complete(TInstance thisRef, Func`3 endMethod, IAsyncResult asyncResult, Boolean requiresSynchronization)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at AMWD.Modbus.Tcp.Client.ModbusClient.<SendRequest>d__54.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at AMWD.Modbus.Tcp.Client.ModbusClient.<ReadHoldingRegisters>d__45.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at SystemMonitor.SystemMonitor.<OnTimedEvent>d__11.MoveNext() in C:\sandbox\utilities\SystemMonitor\SystemMonitor\SystemMonitor.cs:line 62

```

I can not get the exception by adding a call to modbusTCP.Dispose() in the handler for OnDisconnect, but the program hangs on the dispose call.  I have the read inside a an if (modbusTCP.IsConnected), but it doesn't help.  I suspect that the object is getting disposed while the call is timing out.

AndreasAmMueller, 2019-06-09:
Thank You for the report. Will check it soon.

AndreasAmMueller, 2019-07-15:
I've updated the NuGet package with a preview: [0.9.3-master.3](https://www.nuget.org/packages/AMWD.Modbus.Tcp/0.9.3-master.3).

I was not able to reproduce your issue with this new version. Please update and review :-)

AndreasAmMueller, 2019-07-24:
Seems to be closed. Please re-open if you can still reproduce.

closed 2019-07-24 by AndreasAmMueller

underjack, 2019-07-26:
Sorry for the delay, took me a bit to roll back around to this app.

I did some testing, and there's not exception thrown.  However, it raises the disconnected event, when it appears to recreate the connection several times, in the background.  These connection attempts can make it difficult to reconnect when the device is put back on the network.

I am just expecting that if the object has told me that it is disconnected, it should stay disconnected and not be opening ports in the background with orphaned SYN packets.  I mark the connection as closed, dispose of the object, and enable the "connect" button, so that the user can re-connect manually.  If the object is going to attempt re connection on it's own, disconnect should be raised until it gives up, IMO.