AndreasAmMueller/Modbus #8: Modbus TCP/IP server throws in case of an invalid request frame

hmarius, 2020-02-12:
In case of an invalid request frame like: 
"EC 4C 00 00 00 1F 01 17 00 00 00 01 00 00 00 0A 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"
where function code "1F" is invalid, the server will close with the following exception:
Unhandled Exception: System.NotImplementedException: The method or operation is not implemented.
   at AMWD.Modbus.Tcp.Protocol.Request.Deserialize(Byte[] bytes) in C:\tmp\Modbus\src\Modbus.Tcp\Protocol\Request.cs:line 230
   at AMWD.Modbus.Tcp.Protocol.Request..ctor(Byte[] bytes) in C:\tmp\Modbus\src\Modbus.Tcp\Protocol\Request.cs:line 31
   at AMWD.Modbus.Tcp.Server.ModbusServer.HandleClient(TcpClient client) in C:\tmp\Modbus\src\Modbus.Tcp\Server\ModbusServer.cs:line 423
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ThreadPoolWorkQueue.Dispatch()

Maybe at line 445 in /Modbus.Tcp/Server/ModbusServer.cs a catch for NotImplementedException  should be added?

AndreasAmMueller, 2020-02-14:
I see... will investigate and fix it soon. Thank you for your feedback :-)

AndreasAmMueller, 2020-11-28:
Should be fixed with v1.0.0, otherwise reopen.

closed 2020-11-28 by AndreasAmMueller