AndreasAmMueller/Modbus #9: Modbus TCP/IP server keeps spinning in a loop after the client disconnects

hmarius, 2020-02-16:
After the client disconnects the HandleClient method in ModbusTcpServer keeps looping with high cpu usage.
Please see the following test case: https://github.com/hmarius/ModbusTestCase

The problem may be at those lines:
```
var count = await stream.ReadAsync(buffer, 0, buffer.Length);
requestBytes.AddRange(buffer.Take(count));
```
Note: [link](https://stackoverflow.com/questions/34685670/tcpclient-networkstream-readasync-c-sharp)

Thank you!

AndreasAmMueller, 2020-02-17:
I see, thank you for that Link. Will fix that :-)

hmarius, 2020-02-17:
That will be great!
While you are at it :) please take in consideration the case when the received packet is incomplete like in this situation:
[image]
The incomplete packet is: 00 5C 00 00 00 06 01 03 00 00 00 [0a missing]

AndreasAmMueller, 2020-11-28:
Should be fixed with v1.0.0, otherwise reopen.

closed 2020-11-28 by AndreasAmMueller