AndreasAmMueller/Modbus #12: ModbusClientRtu ReadTimeout has no effect

burgerkingorama, 2020-05-14:
When reading Registers from a non-existing (or replying) device you'll never get the timeout exception.

It seems to be a known issue, that when using the ReadAsync methods of the BaseStream the ReadTimeout property is ignored.
I found a discussion [here](https://github.com/dotnet/runtime/issues/28968)

AndreasAmMueller, 2020-05-14:
Thank you for that information. Will update my code with a workaround.

Not sure... this or next weekend... we'll see  ;-)

AndreasAmMueller, 2020-06-18:
The timeouts will be forced here:
- [ReadTimeout](https://github.com/AndreasAmMueller/Modbus/blob/a6d11080c2f5a1205681c881f3ba163d2ac84a1f/src/Modbus.Serial/Util/Extensions.cs#L75)
- [WriteTimeout](https://github.com/AndreasAmMueller/Modbus/blob/a6d11080c2f5a1205681c881f3ba163d2ac84a1f/src/Modbus.Serial/Util/Extensions.cs#L109)

This should fix your problem.

NuGet update coming soon. Need to check whether workaround is also needed on *nix systems.

AndreasAmMueller, 2020-06-24:
Available with [v0.9.8](https://www.nuget.org/packages/AMWD.Modbus.Serial/0.9.8), closing.

Feel free to reopen if you think there might be still a problem.

closed 2020-06-24 by AndreasAmMueller