AndreasAmMueller/Modbus #14: Null reference error in serial client

bdube, 2021-01-07:
Hello,

Thank you for your work on this library.

I have a use case that sometimes triggers a null reference error on the line `reconnectTcs.TrySetResult(true);`
[here](https://github.com/AndreasAmMueller/Modbus/blob/v1.0.2/src/Modbus.Serial/Client/ModbusClient.cs#L1206)
. I temporarily changed the line to `reconnectTcs?.TrySetResult(true);` and I don't see the issue when I make this change.

What I'm trying to do is "discover" which serial port is connected to Modbus by attempting to read a holding register on each available serial port. So in this use case I'm triggering many expected `System.InvalidOperationException` and `System.IO.IOException`, which I think the client sees as a timeout, and so the client tries to reconnect.

Is there a way to prevent automatic reconnection attempts or something else I'm doing wrong that can cause the null reference error? Or possibly a different way to accomplish "port discovery" with the serial client? The serial client has been working well for us in the normal case of connecting immediately to a known serial port, but the port to use is not always known in advance in our use case.

Thank you again for your work on the library.

AndreasAmMueller, 2021-01-11:
Port discovery on serial is a difficult task and I'm thinking of that a lot myself (for an application at work).

I'll check the NULL reference error soon.    
Preventing an auto-reconnect is currently not intended, but may be considered in a future version.

bdube, 2021-01-11:
Thank you for your response.

There is one more line I needed to change to get serial port discovery to work without deadlock. Line 308 in src/Modbus.Serial/Client/ModbusClient.cs:
```
- await Task.WhenAny(ConnectingTask, Task.Delay(Timeout.Infinite, cancellationToken));
+ await Task.WhenAny(ConnectingTask, Task.Delay(ReconnectTimeSpan, cancellationToken));
```

And in the client I use for discovery, I set `ReconnectTimeSpan = TimeSpan.Zero`.

AndreasAmMueller, 2021-03-29:
Should be fixed in [v1.0.3](https://github.com/AndreasAmMueller/Modbus/releases/tag/v1.0.3)

closed 2021-03-29 by AndreasAmMueller