AndreasAmMueller/Modbus #15: Endianness is wrong when converting list of registers into string

mbalous, 2021-03-30:
I have following code

```csharp
using AMWD.Modbus.Common.Util;
using AMWD.Modbus.Tcp.Client;

using (var client = new ModbusClient("10.1.1.2", 8000))
{

    await client.Connect();
    var data = await client.ReadHoldingRegisters(1, 2000, 7);
    var dataString = AMWD.Modbus.Common.Util.Extensions.GetString(data, data.Count, 0, Encoding.ASCII);
    dataString.Dump();
}
```

This code reads 7 registers out of the modbus device and after converting it into string using the `GetString` built-in function produces following result: 
`ZVL--L12004384`
However this result is wrong, because the correct result is following:
`VZ-LL-21003448`

Is there a way to fix this or is the functionality not present in the library?

Thank you

AndreasAmMueller, 2021-04-06:
Ahm... ok... This might happen when the remote device inverts some bytes... Modbus is defined in NetworkOrder (MSB) but many devices might have a LSB order...

I'll add a parameter to flip the bytes - so you can define how the bytes should be interpreted.

EcTomBot, 2021-04-07:
I have a similar problem also reading and writing uint and string values
For example the text in the device is ABCD and the library reads BADC.
When will the parameter for inversion be available?
Thanks

AndreasAmMueller, 2021-04-09:
I'll try to implement it within this weekend.

AndreasAmMueller, 2021-04-13:
Sorry for the delay, now implemented in [v1.0.4](https://github.com/AndreasAmMueller/Modbus/releases/tag/v1.0.4)

```cs
using AMWD.Modbus.Common.Util;
using AMWD.Modbus.Tcp.Client;
using AMWD.Modbus.Common.Util.Extensions;

using (var client = new ModbusClient("10.1.1.2", 8000))
{
    await client.Connect();
    var holdingRegisters = await client.ReadHoldingRegisters(1, 2000, 7);
    string dataString = holdingRegisters.GetString(holdingRegisters.Count, encoding: Encoding.ASCII, flipBytes: true);
    Console.WriteLine(dataString);
}
```

closed 2021-04-13 by AndreasAmMueller

EcTomBot, 2021-04-14:
Hi Andreas.
Thanks for the speed with which you tried to solve the problem, but I believe that the flipbytes parameter must have been provided for all Get methods and in creating registers for writing.
I am not an expert in the MODBUS protocol and I have encountered the problem with a Siemens PLC which expects, for example, that the bytes of the DInt (double word) values are inverted both in reading and writing.
Perhaps the option you have provided is best implemented in the client class and used in all the Get extensions and the Register.Create constructor.
I don't know how much effort this modification takes, but in our case the library as it is is not fully usable.
Thanks again for the help.
Tomaso

AndreasAmMueller, 2021-04-14:
Ahm... okay... I use this library for years now in a major project at work with several manufacturers and never had that issue. *scratch

The Modbus protocol itself is defined in Big-endian ([Spec](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) page 5, 4.2) and as you can see [e.g. here](https://github.com/AndreasAmMueller/Modbus/blob/cb75e55976ff14b8e76179524982cf36697cb541/src/Modbus.Common/Util/Extensions.cs#L97) I'm checking whether it has to be converted first before the system can handle the bytes.

Is it possible to send me a datasheet / manual of the PLC?

mbalous, 2021-04-15:
> Is it possible to send me a datasheet / manual of the PLC?

I'm using the library to connect to Weintek MT8050iE. The datasheet is very non-informative though.
https://dl.weintek.com/public/MT8000iE/eng/Datasheet/MT8050iE1_Datasheet_ENG.pdf

Luke092, 2021-04-15:
Hi Andreas,
me and my colleague @EcTomBot figured out that our Siemens PLC flips the registers for multi-registry data types.
We discovered that there's no standard for the endianness for data that spans across multiple registers as you can see in this article from NI: https://www.ni.com/it-it/innovations/white-papers/14/the-modbus-protocol-in-depth.html.
In our case the PLC stores the data in Little Endian order in term of registers so we needed a way to  write and read data from this kind of situation.

We come out with  an additional flag added to your Extension methods in both reading and writing data for the data types that are longer than 1 register (16bit): int, uint, long, ulong, float and double.

We also added a lenth parameter in the method that prepares the ModbusObject array for writing purposes because in our case the string data are 30 characters long, and every time we want to write a string we need to pad the entire lenght with zeros to clean the registers.

I forked your repository and made a pull request to add this functionalities to your library if you like :)
What i haven't done is to change the Register class to make the above changes in the static Create methods.

Thanks a lot for your work!

Greetings.
Luca