AndreasAmMueller/Modbus #19: Disconnects during reads blocks client

basilfx, 2021-06-17:
I have an application that is basically like this:

```csharp
using var client = new ModbusClient(_host, _port, _logger);

client.ReconnectTimeSpan = ConnectionTimeout;

await client.Connect();

using var cts = new CancellationTokenSource(TimeSpan.FromSeconds(30));

while (true)
{
    var registers = await client.ReadInputRegisters(unit, block.AddressStart, count, cts.Token);

    await Task.Delay(TimeSpan.FromSeconds(30));
}
```

What I observe, is that when the connection does not respond, the `ReadInputRegisters` never returns, even though I am passing a cancellation token. This is the logging:

```
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Connecting with Modbus server at 127.0.0.1:10502.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.Connect enter
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Modbus client starting.
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Modbus client started.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.Reconnect enter
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Connect starting.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.Connect leave
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Reconnected successfully.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.Reconnect leave
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Connected to the Modbus server.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.ReceiveLoop enter
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Receiving responses started.
Exception thrown: 'System.OperationCanceledException' in System.Private.CoreLib.dll
...
Exception thrown: 'System.OperationCanceledException' in System.Private.CoreLib.dll
dbug: MyApp.Connector.Modbus.ModbusConnector[0]
      Time due, processing data.                         <-- This is where my code performs a read.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.ReadInputRegisters enter
dbug: MyApp.Connector.Modbus.ModbusConnector[0]
      Reading input registers from device #1 starting on 0 for 64 registers.
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.SendRequest enter
dbug: MyApp.Connector.Modbus.ModbusConnector[0]
      Sending Request#0 | Device#1, Fn: ReadInputRegisters, Address: 0, Count: 64 |
dbug: MyApp.Connector.Modbus.ModbusConnector[0]
      Request for transaction #0 sent.
Exception thrown: 'System.OperationCanceledException' in System.Private.CoreLib.dll
...
Exception thrown: 'System.OperationCanceledException' in System.Private.CoreLib.dll
```

When I interrupt the application, I get the following additional logging (the application does not exist, nor does it 'continue'):

```
Exception thrown: 'System.IO.EndOfStreamException' in AMWD.Modbus.Tcp.dll
Exception thrown: 'System.IO.EndOfStreamException' in System.Private.CoreLib.dll
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.Reconnect enter
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Reconnect starting.
Exception thrown: 'System.Threading.Tasks.TaskCanceledException' in System.Private.CoreLib.dll
Exception thrown: 'System.Net.Sockets.SocketException' in System.Private.CoreLib.dll
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.SendRequest leave
Exception thrown: 'System.Net.Sockets.SocketException' in System.Private.CoreLib.dll
warn: MyApp.Connector.Modbus.ModbusConnector[0]
      Reading input registers failed: Connection timed out, reconnecting.
System.Net.Sockets.SocketException (110): Connection timed out
   at AMWD.Modbus.Tcp.Client.ModbusClient.SendRequest(Request request, CancellationToken cancellationToken) in /src/lib/Modbus/src/Modbus.Tcp/Client/ModbusClient.cs:line 1345
   at AMWD.Modbus.Tcp.Client.ModbusClient.ReadInputRegisters(Byte deviceId, UInt16 startAddress, UInt16 count, CancellationToken cancellationToken) in /src/lib/Modbus/src/Modbus.Tcp/Client/ModbusClient.cs:line 550
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.ReadInputRegisters leave
Exception thrown: 'System.ArgumentNullException' in AMWD.Modbus.Common.dll
Exception thrown: 'System.ArgumentNullException' in System.Private.CoreLib.dll
trce: MyApp.Connector.Modbus.ModbusConnector[0]
      ModbusClient.Disconnect enter
info: MyApp.Connector.Modbus.ModbusConnector[0]
      Modbus client stopping.
```

To create a server that does not respond, I simply use `socat` (on port 10502):

```
socat TCP4-LISTEN:10502,fork,reuseaddr -
```

One might wonder why I use `socat`: that's because I'm debugging an issue in a cloud application, where a ModBus TCP server gets restarted while an existing connection doesn't seem to get terminated from the client's perspective. The read blocks forever.

I think there are two problems:

- A non-responding or incorrectly responding server should terminate the connection.
- A read (with a CTS) should unblock if the read fails or the CTS is cancelled.

basilfx, 2021-06-17:
I observed the following:

* `ReadHoldingRegisters` or `ReadInputRegisters` invoke `SendRequest`
* `SendRequest` creates a `TaskCompletionSource<Response>`. This instance is stored in `awaitingResponse`. The method then [waits](https://github.com/AndreasAmMueller/Modbus/blob/master/src/Modbus.Tcp/Client/ModbusClient.cs#L1345) for the `TaskCompletionSource<Response>` to complete.
* Meanwhile, the `ReceiveLoop` continuously times-out [here](https://github.com/AndreasAmMueller/Modbus/blob/master/src/Modbus.Tcp/Client/ModbusClient.cs#L1204), because the `socat` server I use does not respond. It then restarts the [while-loop](https://github.com/AndreasAmMueller/Modbus/blob/master/src/Modbus.Tcp/Client/ModbusClient.cs#L1195), meaning that the `TaskCompletionSource<Response>` in `awaitingResponse` is [never](https://github.com/AndreasAmMueller/Modbus/blob/master/src/Modbus.Tcp/Client/ModbusClient.cs#L1224) handled.

basilfx, 2021-06-17:
I modified the [ReceiveLoop](https://github.com/AndreasAmMueller/Modbus/blob/master/src/Modbus.Tcp/Client/ModbusClient.cs#L1215-L1218) a bit:

```csharp
try
{
    byte[] header = await stream.ReadExpectedBytes(6, cts.Token);

    // ...
}
catch (OperationCanceledException) when (timeCts.IsCancellationRequested)
{
    // Cancel all waiting tasks because remote did not respond in time.
    var ex = new TimeoutException("Timeout while waiting for response.");

    foreach (var key in awaitingResponses.Keys)
    {
        if (awaitingResponses.TryRemove(key, out var value))
        {
            value.SetException(ex);
        }
    }

    continue;
}
```

That seems to work: the exception is propagated to the `SendRequest` and eventually to the caller.

I believe this situation is different from a disconnect, where simply reconnecting would work. But I don't know if this is the best solution.

AndreasAmMueller, 2021-06-17:
Hi, thank you for that issue.

We're experiencing a similar problem at work and I'll trace that behaviour very soon, promise.... maybe I can spend a bit of work-time here. -.-'

basilfx, 2021-07-02:
The [above](https://github.com/AndreasAmMueller/Modbus/issues/19#issuecomment-863332765) "workaround" is holding just fine in production, for almost two weeks.

AndreasAmMueller, 2021-07-08:
I've implemented a version to fix some other things around too (CancellationToken of passed into the method should also work on every point).

Should work now in v1.2,0

When there is a problem, please reopen ;-)

closed 2021-07-08 by AndreasAmMueller