AndreasAmMueller/Modbus #20: System.NullReferenceException at AMWD.Modbus.Tcp.Client.ModbusClient.ReceiveLoop()

dcetvericov, 2021-10-05:
First of all, thanks for the nice Modbus library! :-)

Unfortunately, I'm facing a System.NullReferenceException. It happens sporadically, I don't see any system behind these exceptions. All that I can tell you is that it happens in the ReceiveLoop of the ModbusClient.

Stack Trace says:
<body>

  | Name | Value | Type
-- | -- | -- | --
  | StackTrace | "   at AMWD.Modbus.Tcp.Client.ModbusClient.<ReceiveLoop>d__77.MoveNext()" | string

</body>

It occurs when using v1.2 of the NuGet package.

AndreasAmMueller, 2021-10-05:
May you have some additional information on your environment?
- requested method (with parameters)
- interval
- devices on that IP address
- ...

Maybe I can reproduce this problem somehow.

dcetvericov, 2021-10-14:
Sorry for not answering for such a long time. Unfortunately, I must deal with some more important stuff not using your MODBUS library at the moment. I'll let you know more as soon as I face these issues again.

gnampf89, 2022-02-21:
I think I have the same error and my guess is that this happens when the response is already received when the request is still being sent (see log). And after the timeout this leads to a TaskCanceledException. FrmModbusTest is log from my code, ModbusClient is log from your code. Timeout in this example is 5Seconds. 

`2022-02-21 16:10:22,365 DEBUG FrmModbusTest [(null)] - >SendCmd()
2022-02-21 16:10:22,365 DEBUG ModbusClient [(null)] - ModbusClient.WriteSingleRegister enter
2022-02-21 16:10:22,367 DEBUG ModbusClient [(null)] - Writing a register to device #1: Holding Register #16 | Hi: 00 Lo: 47 | 71.
2022-02-21 16:10:22,367 DEBUG ModbusClient [(null)] - ModbusClient.SendRequest enter
2022-02-21 16:10:22,367 DEBUG ModbusClient [(null)] - Added transaction #4 to receive queue
2022-02-21 16:10:22,368 DEBUG ModbusClient [(null)] - Sending Request#4 | Device#1, Fn: WriteSingleRegister, Address: 16, Count: 0 | 00 47
2022-02-21 16:10:22,433 DEBUG ModbusClient [(null)] - Received response for transaction #4.
2022-02-21 16:10:22,472 ERROR ModbusClient [(null)] - Unexpected error (NullReferenceException) on receive: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
System.NullReferenceException: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
   bei AMWD.Modbus.Tcp.Client.ModbusClient.<ReceiveLoop>d__77.MoveNext()
2022-02-21 16:10:22,512 DEBUG ModbusClient [(null)] - Request for transaction #4 sent
2022-02-21 16:10:27,590 DEBUG ModbusClient [(null)] - ModbusClient.SendRequest leave
2022-02-21 16:10:27,627 DEBUG ModbusClient [(null)] - ModbusClient.WriteSingleRegister leave
2022-02-21 16:10:27,758 ERROR FrmModbusTest [(null)] - SendCmd():
System.Threading.Tasks.TaskCanceledException: Eine Aufgabe wurde abgebrochen.
...
`

leoyordi, 2022-02-24:
First, thank you for the lib.
I've same problem.
Function for read Coils
```
        public bool[] LerCoil(int posicao, int matriz)
        {
            if (modbus == null) return null;
            try
            {
                ushort p = (ushort)posicao;
                ushort m = (ushort)matriz;
                var coil = modbus.ReadCoils(_deviceID, p, m).Result;
                if (coil == null || coil.Count == 0) return null;
                return coil.Select(m => m.BoolValue).ToArray();
            }
            catch (SocketException) { }
            catch (NullReferenceException) { }
            catch (ObjectDisposedException) { }
            catch (Exception e)
            {
                Error($"Lendo [{posicao},{matriz}]: {e.Message}");
            }
            return null;
        }
```
Function for read HoldingRegisters is similar but
`var list = modbus.ReadHoldingRegisters(_deviceID, p, m).Result;`
My log:
```
10:12:09 [ERR] Unexpected error (NullReferenceException) on receive: Object reference not set to an instance of an object.
System.NullReferenceException: Object reference not set to an instance of an object.  at AMWD.Modbus.Tcp.Client.ModbusClient.ReceiveLoop()
10:12:29 [ERR] [LerCoil] Lendo [260,1]: One or more errors occurred. (A task was canceled.)
```
I do a constant loop.
1st I read a large sequence of registers and coils with functions above.
2nd I analyze each of the variables. Depending on the scenario, I write in some variables (functions not shown here) and try to confirm that the writing worked by re-reading the value using the above functions.
3rd Return to the beginning of the loop.

If I have user action, I also use the above functions for reading and the same (not mentioned here) for writing. These actions are concurrent with the loop above.

Is it the case to use some data flow control?

taai, 2022-05-14:
@gnampf89 So, your fork is working just fine? I'm thinking about what to do now, as it looks like that the author has abandoned the project...

gnampf89, 2022-05-14:
Yes, works for me at least in my test environment. Tested with the real hardware I need, but no 24/7 operation so far.

AndreasAmMueller, 2022-05-15:
Sorry for letting you down guys. I'm currently very busy at work and my honorary job at the rescue service does the rest (preparing for some big events)... I didn't forget you but just don't have the time 😢

taai, 2022-05-15:
@AndreasAmMueller I'm not judging. It is free and opensource software – nobody expects you to always work on it, it is not your day job. 😉 I'm glad that you are ok (you never know) and that you are just busy. I think it is awesome that you are working for the rescue service. Helping people is very honorable work. It is not your duty to maintain your code. Do what makes you happy! 🤗

graineeg, 2022-06-22:
Hi all.I found that nullreference exception is ocuring in RecieveLoop() [line 1283](https://github.com/AndreasAmMueller/Modbus/blob/32e169fa54a1e59168e501306607a01eb65bfa5c/src/Modbus.Tcp/Client/ModbusClient.cs#L1283)
```C#
if (queueItem != null)
        {
		if (!DisableTransactionId)
			logger?.LogDebug($"Received response for transaction #{response.TransactionId}.");

		queueItem.CancellationTokenSource.Dispose(); // CancellationTokenSource is null - Dirty fix: ?.Dispose();
		queueItem.TaskCompletionSource.TrySetResult(response);
		queueItem.TimeoutCancellationTokenSource.Dispose(); // TimeoutCancellationTokenSource is null - Dirty fix: ?.Dispose();
	}
```
 when QueuedRequest queueItem is not fully initialized as @gnampf89 wrote.

closed 2024-04-02 by AndreasAmMueller