AndreasAmMueller/Modbus #21: client.Dispose() blocks

RolfDieterClavery, 2021-12-20:
I am trying to test features of several Modbus device using the TCP client in a loop. For each IP address a new client is instantiated:
```
using (var client = new ModbusClient(availableHost, port, logger))
{
    await client.Connect();
    var resultDict = await client.ReadDeviceInformation(device, AMWD.Modbus.Common.DeviceIDCategory.Basic);
    // evaluate data if available
    await client.Connect();
    var registers = await client.ReadHoldingRegisters(1, 10, 1);
    // evaluate data if available
}
```
Unfortunately one/some of the Modbus devives reject the connect as they allow only one connection. The result are failing calls `ReadDeviceInformation()` and `ReadHoldingRegisters`. This is not a problem. But at least `ReadHoldinRegisters` often throws an exception (A Task was canceled).

Handling the exception inside or outside the using block the disposing of the ModbusClient when leaving the using block or explicitely called does not return. None of the timeouts the ModbusClient provides limits the time of disposing the client.

Debugging the code as far as possible points out a problem during Dispose()/Disconnect() in the ReceiveLoop (line 1334) and Reconnect (line 1121).

enricobenedos, 2022-04-21:
We are using this library in production and sometimes some of these devices are not reachable from network. In case of device is not reachable the library has a very strange behaviour:
- The `connect()` method goes in timeout but after it the execution simply continue without any exception. Why not simply generate connection exception? (Is it a fake `connect()`?)
- At this point the code reach a read data method as for example `ReadHoldingRegisters` that generates exception
- Then the client need to be disposed but the `dispose()` method does not return, so the application is stucked

xiangzLL, 2022-11-05:
why the client need to reconnect when ReadHoldingRegisters face a timeout exception, most of the time it just the device is offline?

closed 2024-04-02 by AndreasAmMueller